jamesturton/bose-qc35-usb-c #3: Add information about some BOM items

choumarin, 2022-04-10:
Hi,

Great project, going to try and make this happen.
The BOM is missing information on resistors:
- R1, R2, R3, R4, R17, R18, R19 is missing a value
- R5, R21 is indicating 0R with a *

According to the picture, all of those those seems to be DNP, is that correct?

Also, do all the LEDs have the same color?

Thanks!

jamesturton, 2022-04-19:
Hi @choumarin!
I've now updated the schematic (2d783980ca84bd099cc5140e28db72a5c78a2f65) with values R1, R2, R3, R4, R17, R18, R19.
They provide ESD protection for the MCU, so are actually quite important if you want use the headphones outside of an ESD controlled lab.
R5, R21 are alternatives for Q1 and U1 respectively - see the comments on the schematic for more detail.
I have also given example LED part numbers for parts with suitable brightness.

All the best building this!
James

closed 2022-04-19 by jamesturton

choumarin, 2022-06-12:
Finally done, I had to borrow some equipment to solder that small, especially J1.
[image]

Thanks so much for this great project!

You don't have discussion pages, so I'm just gonna add stuff here:
- I measured LED_POWER_ON at 3.8V so I decided on using the 5V ESD components (you have a comment saying maybe need 3.3V)
- couldn't find a replacement for U1 unfortunately, so using R21
- This was my BOM at digikey, with some replacement of the original BOM, it might help other people to find replacements:

| QUANTITY SHIPPED | PART NUMBER                       | DESCRIPTION                      | CUSTOMER REFERENCE            |
|------------------|-----------------------------------|----------------------------------|-------------------------------|
|               30 | CG0402MLC-05LGCT-ND               | VARISTOR 0402                    | R1, R2, R3, R4, R17, R18, R19 |
|               10 | 3757-BAT54CW_R1_00001CT-ND        | SOT-323, SKY                     | D5                            |
|               10 | 160-1832-1-ND                     | LED GREEN CLEAR CHIP SMD         | D1                            |
|               10 | 3147-B1911USD-20D000114U1930CT-ND | LED RED CLEAR 1608 SMD           | D2                            |
|               10 | 475-2831-1-ND                     | LED COOL WHITE DIFFUSED 0603 SMD | D3                            |
|               10 | 1830-1061-1-ND                    | LED BLUE CLEAR 0603 SMD          | D4                            |
|               10 | 23-5024262030CT-ND                | CONN RCPT 20POS SMD GOLD         | J1                            |
|               10 | WM12846CT-ND                      | CONN FFC FPC 5POS 0.50MM R/A     | J3                            |
|               10 | 900-2171790001CT-ND               | CONN RCP USB2.0 TYP C 24P SMD RA | J2                            |
|               10 | 490-11022-1-ND                    | FERRITE BEAD 330 OHM 0402 1LN    | FB1                           |
|               30 | 1292-WR04X1001FTLCT-ND            | RES 1K OHM 1% 1/16W 0402         | R11, R14, R15, R16            |
|               10 | RMCF0402JT1K20CT-ND               | RES 1.2K OHM 5% 1/16W 0402       | R9                            |
|               10 | CR0402-JW-103GLFCT-ND             | RES SMD 10K OHM 5% 1/16W 0402    | R6, R20                       |
|               10 | RMCF0402JT5K10CT-ND               | RES 5.1K OHM 5% 1/16W 0402       | R12, R13                      |
|               10 | RMCF0402JT7K50CT-ND               | RES 7.5K OHM 5% 1/16W 0402       | R7                            |
|               10 | RMCF0402JT3K90CT-ND               | RES 3.9K OHM 5% 1/16W 0402       | R8                            |
|               10 | CR0402-FX-5100GLFCT-ND            | RES SMD 510 OHM 1% 1/16W 0402    | R10                           |
|               10 | 1292-WR04X2200FTLCT-ND            | RES 220 OHM 1% 1/16W 0402        | R22                           |
|               10 | RMCF0402ZT0R00CT-ND               | RES 0 OHM JUMPER 1/16W 0402      | R21, R5                       |
|               10 | 1276-1010-1-ND                    | CAP CER 1UF 6.3V X5R 0402        | C2                            |
|               10 | 1276-1482-1-ND                    | CAP CER 4.7UF 10V X5R 0402       | C1                            |
|               10 | 2SC5658T2LSCT-ND                  | TRANS NPN 50V 0.15A VMT3         | Q1                            |
|               30 | CG0402MLC-3.3LGCT-ND              | VARISTOR 0402                    | R1, R2, R3, R4, R17, R18, R19 |

RavaszEmber, 2023-03-30:
This is my first real SMD project so bear with me if I'm asking any stupid questions.

- I just bought 5 PCB's from a manufacturer without the PCB stencil or PCB assembly (hope I can assemble this myself, I know I'll be kicking myself later for this).
- I bought a 0402 resistor sample book to cover all the resistors
- I also bought a set of 0402 capacitor tapes for the two caps C1 and C2
- I followed @choumarin's Digikey BOM for the other components and found equivalents where needed. My Digikey cart is as follows: 

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/benja/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/benja/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

Index | Quantity | Part Number | Manufacturer Part Number | Description | Customer Reference | Available | Backorder | Unit Price | Extended Price USD
-- | -- | -- | -- | -- | -- | -- | -- | -- | --
1 | 5 | 23-5024262030CT-ND | 5.02E+09 | CONN RCPT 20POS SMD GOLD | J1 | 5 | 0 | 1.42 | 7.1
2 | 5 | 900-2171790001CT-ND | 2.17E+09 | CONN RCP USB2.0 TYP C 24P SMD RA | J2 | 5 | 0 | 0.9 | 4.5
3 | 5 | 900-0512810598CT-ND | 5.13E+08 | CONN FFC FPC 5POS 0.5MM R/A | J3 | 5 | 0 | 0.51 | 2.55
4 | 5 | 490-11022-1-ND | BLM15PX331SN1D | FERRITE BEAD 330 OHM 0402 1LN | FB1 | 5 | 0 | 0.1 | 0.5
5 | 5 | 2SC5658T2LQCT-ND | 2SC5658T2LQ | TRANS NPN 50V 0.15A VMT3 | Q1 | 5 | 0 | 0.4 | 2
6 | 5 | 3757-BAT54CW_R1_00001CT-ND | BAT54CW_R1_00001 | SOT-323, SKY | D5 | 5 | 0 | 0.21 | 1.05

</body>

</html>

Am I missing anything here? I noticed U1 was missing but saw your above comment that R21 was an alternative to this (may be easier to just order U1 and not need it than need it and not have it).
I'm assuming I add everything to the PCB when assembling or am I mistaken? Are some components not necessary?

Thanks!

koolaidxk1d, 2024-03-09:
> This is my first real SMD project so bear with me if I'm asking any stupid questions.
> 
>     * I just bought 5 PCB's from a manufacturer without the PCB stencil or PCB assembly (hope I can assemble this myself, I know I'll be kicking myself later for this).
> 
>     * I bought a 0402 resistor sample book to cover all the resistors
> 
>     * I also bought a set of 0402 capacitor tapes for the two caps C1 and C2
> 
>     * I followed @choumarin's Digikey BOM for the other components and found equivalents where needed. My Digikey cart is as follows:
> 
> 
> <style> </style>
> Index 	Quantity 	Part Number 	Manufacturer Part Number 	Description 	Customer Reference 	Available 	Backorder 	Unit Price 	Extended Price USD
> 1 	5 	23-5024262030CT-ND 	5.02E+09 	CONN RCPT 20POS SMD GOLD 	J1 	5 	0 	1.42 	7.1
> 2 	5 	900-2171790001CT-ND 	2.17E+09 	CONN RCP USB2.0 TYP C 24P SMD RA 	J2 	5 	0 	0.9 	4.5
> 3 	5 	900-0512810598CT-ND 	5.13E+08 	CONN FFC FPC 5POS 0.5MM R/A 	J3 	5 	0 	0.51 	2.55
> 4 	5 	490-11022-1-ND 	BLM15PX331SN1D 	FERRITE BEAD 330 OHM 0402 1LN 	FB1 	5 	0 	0.1 	0.5
> 5 	5 	2SC5658T2LQCT-ND 	2SC5658T2LQ 	TRANS NPN 50V 0.15A VMT3 	Q1 	5 	0 	0.4 	2
> 6 	5 	3757-BAT54CW_R1_00001CT-ND 	BAT54CW_R1_00001 	SOT-323, SKY 	D5 	5 	0 	0.21 	1.05
> 
> Am I missing anything here? I noticed U1 was missing but saw your above comment that R21 was an alternative to this (may be easier to just order U1 and not need it than need it and not have it). I'm assuming I add everything to the PCB when assembling or am I mistaken? Are some components not necessary?
> 
> Thanks!

Did you ever complete this?

amomchilov, 2024-11-21:
@RavaszEmber Thanks for sharing that list. For future readers, here's a CSV that you can import into Digikey. It only includes the 5V varistors rather than the 3.3V.

```csv
Country of Origin may be different at time of shipment.
"The HTSUS and ECCN information shown is for informational purposes only and is not a representation or warranty as to the accuracy or reliability of these classifications. All information is provided ""as is"" and is subject to change without notice. Any use made of the information provided is without recourse to Digi-Key and at the user's risk."
Index,Manufacturer Part Number,Manufacturer Name,Description,Availability,Stock Status,Attrition %,Requested Quantity 1,Pack Quantity 1,Pack Type 1,Digi-Key Part Number 1,Unit Price 1,Extended Price 1,Minimum Order Quantity 1,Customer Reference,Requested Part Number,Lead Weeks,Datasheet,Reference Designator,US Import Tariff,Note,Part Status,RoHS Status,Moisture Sensitivity Level (MSL),ECCN,HTSUS,Country of Origin,Environmental Information,Original Part numbers,Image,Product Notes
1,CG0402MLC-05LG,Bourns Inc.,VARISTOR 0402,"49,981",In Stock,,7,7,Cut Tape (CT),CG0402MLC-05LGCT-ND,0.86000,$6.02,1,,CG0402MLC-05LGCT-ND,10,https://www.bourns.com/docs/product-datasheets/cga-mlc.pdf,,	-,,Active,ROHS3 Compliant,1  (Unlimited),EAR99,8533.40.4000,Taiwan,https://www.bourns.com/docs/rohs-content/bourns-reach-letter.pdf|https://www.bourns.com/docs/rohs-cofc/chipguard_mlu_mlc_-coc2.pdf?sfvrsn=ac55b99d_2,CG0402MLC-05LGCT-ND,https://mm.digikey.com/Volume0/opasdata/d220001/medias/images/639/CG0402MLC.jpg,
2,BAT54CW_R1_00001,Panjit International Inc.,"SOT-323, SKY","16,420",In Stock,,1,1,Cut Tape (CT),3757-BAT54CW_R1_00001CT-ND,0.26000,$0.26,1,,3757-BAT54CW_R1_00001CT-ND,8,https://www.panjit.com.tw/upload/datasheet/BAT54W_SERIES.pdf,,	-,,Active,ROHS3 Compliant,1  (Unlimited),EAR99,8541.10.0070,Taiwan,,3757-BAT54CW_R1_00001CT-ND,https://mm.digikey.com/Volume0/opasdata/d220001/medias/images/2791/BAS40CW-AR1XX.jpg,
3,LTST-C193TGKT-5A,Lite-On Inc.,LED GREEN CLEAR CHIP SMD,"112,473",In Stock,,1,1,Cut Tape (CT),160-1832-1-ND,0.78000,$0.78,1,,160-1832-1-ND,22,https://mm.digikey.com/Volume0/opasdata/d220001/medias/docus/1168/LTST-C193TGKT-5A.pdf,,May apply on shipments to the U.S.,,Active,ROHS3 Compliant,2A  (4 Weeks),EAR99,8541.41.0000,Thailand,,160-1832-1-ND,https://mm.digikey.com/Volume0/opasdata/d220001/medias/images/403/LTST-C193TBKT-5A.jpg,
4,B1911USD-20D000114U1930,Harvatek Corporation,LED RED DIFFUSED 0603 SMD,"3,820",In Stock,,1,1,Cut Tape (CT),3147-B1911USD-20D000114U1930CT-ND,0.16000,$0.16,1,,3147-B1911USD-20D000114U1930CT-ND,8,https://mm.digikey.com/Volume0/opasdata/d220001/medias/docus/3694/B1911USD-20D000114U1930.pdf,,	-,,Active,RoHS Compliant,2A  (4 Weeks),EAR99,8541.41.0000,Taiwan,https://mm.digikey.com/Volume0/opasdata/d220001/medias/docus/6223/Harvatek%20Corp%20REACH%20235%20Jun%202023%20Statement.pdf,3147-B1911USD-20D000114U1930CT-ND,https://mm.digikey.com/Volume0/opasdata/d220001/medias/images/4095/MFG_B1911.jpg,"Non Cancellable Non Returnable. "
5,LW Q38E-Q1OO-3K6L-1-5-R18,ams-OSRAM USA INC.,LED COOL WHITE DIFFUSED 0603 SMD,"83,641",In Stock,,1,1,Cut Tape (CT),475-LWQ38E-Q1OO-3K6L-1CT-ND,0.56000,$0.56,1,,475-2831-1-ND,10,https://dammedia.osram.info/media/resource/hires/osram-dam-2493068/LW%20Q38E.pdf,,	-,,Active,ROHS3 Compliant,2  (1 Year),EAR99,8541.41.0000,Taiwan,,475-2831-1-ND,https://mm.digikey.com/Volume0/opasdata/d220001/medias/images/599/LB-Q39E.jpg,
6,IN-S63AT5B,Inolux,LED BLUE CLEAR 0603 SMD,"7,178",In Stock,,1,1,Cut Tape (CT),1830-1061-1-ND,0.39000,$0.39,1,,1830-1061-1-ND,5,https://www.inolux-corp.com/datasheet/SMDLED/Mono%20Color%20Top%20View/IN-S63AT%20Series_V1.4.pdf,,May apply on shipments to the U.S.,,Active,ROHS3 Compliant,3  (168 Hours),EAR99,8541.41.0000,China,,1830-1061-1-ND,https://mm.digikey.com/Volume0/opasdata/d220001/medias/images/778/IN-S63AT.JPG,
7,5024262030,Molex,CONN RCPT 20POS SMD GOLD,"14,379",In Stock,,1,1,Cut Tape (CT),23-5024262030CT-ND,2.28000,$2.28,1,,23-5024262030CT-ND,12,https://www.molex.com/pdm_docs/ps/PS-502426-001.pdf,,	-,,Active,ROHS3 Compliant,1  (Unlimited),EAR99,8536.69.4040,Japan,https://www.molex.com/datasheets/rohspdf/5024262030_rohs.pdf,23-5024262030CT-ND,https://mm.digikey.com/Volume0/opasdata/d220001/medias/images/661/5024262030.JPG,
8,0512810594,Molex,CONN FFC FPC 5POS 0.5MM R/A,,Obsolete,,1,,,,,,,,WM12846CT-ND,14,https://www.molex.com/pdm_docs/sd/512810794_sd.pdf,,	-,,Obsolete,ROHS3 Compliant,1  (Unlimited),EAR99,8536.69.4051,Japan,https://www.molex.com/datasheets/rohspdf/0512810594_rohs.pdf,WM12846CT-ND,https://mm.digikey.com/Volume0/opasdata/d220001/derivates/6/200/264/240/0512810594_web%28640x640%29.jpg,"No Backorders Allowed. "
9,2171790001,Molex,CONN RCP USB2.0 TYP C 24P SMD RA,"3,286",In Stock,,1,1,Cut Tape (CT),900-2171790001CT-ND,1.44000,$1.44,1,,900-2171790001CT-ND,17,https://www.molex.com/webdocs/datasheets/pdf/en-us/2171790001_IO_CONNECTORS.pdf,,May apply on shipments to the U.S.,,Active,ROHS3 Compliant,1  (Unlimited),EAR99,8536.69.4040,China,,900-2171790001CT-ND,https://mm.digikey.com/Volume0/opasdata/d220001/medias/images/2912/MFG_2171790001.jpg,
10,BLM15PX331SN1D,Murata Electronics,FERRITE BEAD 330 OHM 0402 1LN,"225,958",In Stock,,1,1,Cut Tape (CT),490-11022-1-ND,0.16000,$0.16,1,,490-11022-1-ND,17,https://www.murata.com/en-us/products/productdata/8796740059166/ENFA0018.pdf,,	-,,Active,ROHS3 Compliant,1  (Unlimited),EAR99,8548.00.0000,Thailand,,490-11022-1-ND,https://m [... cut ...] 0001/medias/images/2537/Ceramic-Capacitor-CL-Series.jpg,
22,2SC5658T2LS,Rohm Semiconductor,TRANS NPN 50V 0.15A VMT3,"41,096",In Stock,,1,1,Cut Tape (CT),2SC5658T2LSCT-ND,0.42000,$0.42,1,,2SC5658T2LSCT-ND,18,https://fscdn.rohm.com/en/products/databook/datasheet/discrete/transistor/bipolar/2sc4617e3-e.pdf,,	-,,Not For New Designs,ROHS3 Compliant,1  (Unlimited),EAR99,8541.21.0075,South Korea,,2SC5658T2LSCT-ND,https://mm.digikey.com/Volume0/opasdata/d220001/medias/images/4846/846_VMT3%20Pkg.jpg,
```