jamesturton/bose-qc35-usb-c #5: Question about buying parts

ZasoZ, 2022-07-29:
Hi I thought this was a really cool project but I was wondering if you knew any stores that sells all the parts and ships to Sweden ?

Kind regards

Agnus-Castus, 2023-09-11:
https://www.mouser.se or https://www.digikey.se/en

amomchilov, 2023-11-12:
@jamesturton Does anything prevent us from just scrapping the original board for parts, and transplanting them over?

jamesturton, 2023-11-13:
> @jamesturton Does anything prevent us from just scrapping the original board for parts, and transplanting them over?

You can absolutely do this! I tried to layout the circuit board with the components in the same place to make it easier when moving them across. The only thing to note is that you will need the two 5k1 resistors which are required by the USB spec for USB gen2 when using USB type-C. And the new USB type-C port of course!

jonas-stg, 2024-03-20:
Maybe someone could provide direct links for parts?
I don't know how to search for some of them, or could give me an example for 
[image]

closed 2024-09-13 by jamesturton