jamesturton/bose-qc35-usb-c #7: Does this work with QC35 II's

DerZade, 2023-03-23:
In all files it says QC35, but not QC35 **II**.
Does this mod work for QC35 II as well?

jamesturton, 2023-03-23:
Very good question! I don't actually know as I don't have the QC35 II's, so not sure if they changed the design of the board at all, but from what I can see, the right ear cup looks identical between I and II so I think it should work. My wife actually has a pair of the QC35 II's so I could try opening them while she is sleeping some time :laughing:

DerZade, 2023-03-23:
That would be amazing 🙈 There was a guy on reddit who apparently uses this mod with II's, but I'm not 100% sure. I'll probably PM him and ask to clarify.

jamesturton, 2023-03-23:
Great, thanks for the update! 🙂 I shall update the readme

jamesturton, 2023-03-23:
Fixed in  7100c0c!

closed 2023-03-23 by jamesturton