jamesturton/bose-qc35-usb-c #8: BOM update

Edison517, 2023-09-15:
I may have missed it, but I had to DNP R1-5 to get the pcb to work when installed...I only tried that because of the picture you posted of the assembled board...I also was testing the LED orientation and all 4 would light up with R1-5 populated, but then they test correctly, individually when R1-5 are DNP. Just my $0.02. Great work, otherwise! Thanks!

jamesturton, 2023-09-19:
Did you use `CG0402MLC-05LG` for R1-4?
I didn't mount anything in those as you saw :wink: So I've updated the schematic to make this much clearer!

Edison517, 2023-09-19:
no, I didn't take a look at the schematic at all (I don't have KiCad installed), so I didn't see until today that I was supposed to use those varistors and not 0ohm resistors (oops!). I'd like to see a schematic pdf posted ;-) and I ordered those varistors to install later.

closed 2024-09-13 by jamesturton