jamesturton/bose-qc35-usb-c #9: Missing BOM in CSV and CPL (/Centroid) Files for PCB Fab site upload

Shoothimnow, 2023-10-14:
Hello, attempting to send this off to a PCB fabricator (first time ever doing so) and cannot for the life of me fine the CPL file.  Personally also afraid to copy paste the BOM html file into a CSV as mistype anything.  Would love to attempt this modification, but lack the ordering skills and soldering skills to do so by myself it seems.

theo-m, 2023-11-28:
Hey I'm interested in this as well, would love to keep using my headphones with USB-C, but I'm totally incompetent to do the soldering part.

krikkirk, 2024-08-07:
Hi @Shoothimnow I created the CSV and CPL for the USBC connector side, but was quoted over $60 on the JLC website with a new user discount. In my opinion, spending $60 for a half-completed PCB is not economical. You might be much better off selling your old headphones and buying the latest model with a USBC port.

jamesturton, 2024-09-13:
I have heard of many people using the USB boards from the newer QC45 in their QC35 which fits just as good. I don't know of any official sources for the replacement USB boards for the QC45 but I know other have been able to buy them cheaply from China so I suggest you look there.

closed 2024-09-13 by jamesturton