jamesturton/bose-qc35-usb-c #10: Original QC35 Schematic

tpruszin, 2023-12-20:
Hi James could you upload the original schematic for the QC35 aswell?

jamesturton, 2023-12-27:
Hi @tpruszin!
I don't have any original schematics. I drew the schematic in this repository from scratch.

tpruszin, 2023-12-28:
Hey there @jamesturton!
do you have the original board you drew for the original QC35 board? or did you go direct to this design? 
I've successfully mapped as much as I can of the QC45 board and integrated it into my QC35 II's, and it's been working perfectly for a week now. However, I've noticed that one Blue LED remains on continuously when off and high power when headphones are on. I suspect it's linked to the LED_Power_ON line. I'm trying to understand the flow of this line and figure out why this is the case. My goal is to replicate the original board's behavior, potentially a modification like yours that doesnt need to fab a custom PCB but where it doesn't activate anything in its default state.
Thanks!

jamesturton, 2023-12-28:
That sounds cool!
The only thing I changed/added to the original design is to change the USB-C connector and add the two 51R resistors on the CC lines. The rest of the components are the same.

So do you have both a pair of 35s and 45s? Can you post a photo of how the QC45 circuit board looks?

tpruszin, 2023-12-28:
Okay no worries! So I managed to find a seller on AliExpress selling QC45 boards individually! I will eventually try and make this look better but here is my very rough photos and tracing to try and figure out where everything went flowing your diagram at the same time to try and understand things. 
[image]
[image]
[image]
[image]
[image]

tpruszin, 2023-12-28:
[image]
[image]
[image]
[image]
[image]

polyrhythmatic, 2024-01-24:
@tpruszin very cool! This would be great, please keep us updated

polyrhythmatic, 2024-01-24:
@jamesturton wondering your thoughts on instead using a board like this, and wiring it to the micro USB pinouts

https://www.tindie.com/products/nullstalgia/mini-usb-c-5v-breakout-null-c/

jamesturton, 2024-01-25:
> @jamesturton wondering your thoughts on instead using a board like this, and wiring it to the micro USB pinouts
> 
> https://www.tindie.com/products/nullstalgia/mini-usb-c-5v-breakout-null-c/

It's an interesting thought. It looks like that specific USB-C adapter won't work with any C-C cables as it's missing the two resistors on CC1 and CC2. I also probably wouldn't recommend this as it would be very difficult to mechanically secure the connector down to the old PCB if you can't solder it - maybe it works for a few days/weeks, but plugging and unplugging the connector that many times causes a lot of forces at the connector so it would need to be very well secured. I also doubt that there is enough room in the headphones to accommodate the extra PCB as the USB-C connector is already slightly larger than the micro-USB connector.

But if you do decide to try it then please post your results!

closed 2024-09-13 by jamesturton