jamesturton/bose-qc35-usb-c #11: LEDs don't work

DieHertz, 2024-03-04:
Hi, thank you for your effort and for the schematics!

I had asked my friend from China to produce a few boards according to the schematic and BOM, and have successfully installed one today. All of the buttons work, charging works, but none of the LEDs come up, all the boards I have are like that.

I know it's probably some fault on my side, but maybe you've got an idea what could be wrong or how to troubleshoot.

jamesturton, 2024-03-04:
Hi @DieHertz!
The first thing I would do would be to check polarity of the LEDs you are using. Sometimes they have the marking on the anode and sometimes on the cathode.

DieHertz, 2024-03-14:
Finally saw what my guy did wrong - he did not read the instructions on Q1 and U1. I removed some optional components and it still does not work. Back to the bench I guess :-D

closed 2024-03-15 by DieHertz

DieHertz, 2024-03-16:
Somehow the fuse was causing issues with charging (it looked like the headphones were alternating between charging and not charging by rapidly blinking the green LED).
I have discovered so by elimination, when everything else failed, I tried supplying external 5V directly to VCC and it started charging. Then I tried shorting the fuse and it also started charging.

That's weird, no idea what can be wrong with the fuse, but it's a correct 330R one according to my multimeter. All 5 boards behave like that, shorting the fuse making them work properly for charging.