jamesturton/bose-qc35-usb-c #12: BOM component placement questions

rogerthemar, 2024-03-29:
[image]
hello!
i have a question regarding the BOM. i have gone ahead and made a small order of these boards (minimum was 4) 
i pulled one out but it does not seem to work, have i placed components i should not have? Per the photos that you uploaded
[image]
[image]
there seems to be components missing from what your BOM states however when i sent this to the factory they placed all the components. should i be removing some components from the board to get this to work?

rogerthemar, 2024-03-29:
[image]
[image]

rogerthemar, 2024-03-29:
here is the other side of my board. 
[image]

jamesturton, 2024-03-30:
Hi @rogerthemar!
Yes, there should be some components not mounted. If you look at the original board from bose you will be able to see the components that should be missing. I've also written an explanation for the missing components in the schematic so that's well worth looking at too.
Hope this helps!

DieHertz, 2024-03-31:
> Hi @rogerthemar! Yes, there should be some components not mounted. If you look at the original board from bose you will be able to see the components that should be missing. I've also written an [...]

@jamesturton I have figured mine out, thank you for your effort.
I think Q1 is not optional for Bose QC35II as it's an inverter, without it blue LED will stay on when it should not.

rogerthemar, 2024-04-01:
hi, i went ahead and removed the extra components that should not have been placed and everything is now working perfectly. i ended up with a few extra boards due to a minimum order requirement so if i anyone is interested i can send that out for marginal cost! thanks for all the help!

como-stella, 2024-04-06:
> <img alt="asdf" width="410" src="https://private-user-images.githubusercontent.com/165359259/317962869-61874ae8-a298-40a6-aa14-6a1e521ad6e1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTI0NDc1MzUsIm5iZiI6MTcxMjQ0NzIzNSwicGF0aCI6Ii8xNjUzNTkyNTkvMzE3OTYyODY5LTYxODc0YWU4LWEyOTgtNDBhNi1hYTE0LTZhMWU1MjFhZDZlMS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDA2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQwNlQyMzQ3MTVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yODFjNDViNGEwZGI2ZGFhOGJmMGQ2ZGFmY2EzYjRhNjZlMDM0ZDEwNjg3MDcwNWYzNGQwNTBlMjlhZWIxOTBmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.RFhq1m5iOcVqp83GL17a0EXNWtNDaqrYTas0ZZ7jOFE"> hello! i have a question regarding the BOM. i have gone ahead and made a small order of these boards (minimum was 4) i pulled one out but it does not seem to work, have i placed components i should not have? Per the photos that you uploaded
> there seems to be components missing from what your BOM states however when i sent this to the factory they placed all the components. should i be removing some components from the board to get this to work?

Hello! Did you figured out your board to work on qc35 ii?

If you have 3 left overs... may I buy 1 off of you? Let me know!

closed 2024-05-14 by rogerthemar