jamesturton/bose-qc35-usb-c #13: Part number of the new USB C port?

krikkirk, 2024-08-06:
Hey there, I noticed the most recent commit is about a new USB C port without the need of a shell mod. Is it just an new USB C part or there are changes about the PCB as well. Can you upload the new gerber file and the new part number?

jamesturton, 2024-08-07:
The old USB C connector had some plastic support pins which needed removing. So it was the USB C connector its self which required modification. The casing will require cutting even with the new part due to the size of the USB C connector. The new part is also more widely available, rather than just being able to purchase from china.

krikkirk, 2024-08-07:
Hi James, thanks for the explanation, it all makes sense now.

closed 2024-08-07 by krikkirk