jamesturton/bose-qc35-usb-c #18: Bad fit with new USB connector

Kartoffel, 2025-01-18:
The USB-C connector added in fa917dc (Molex 2171790001) connects too deep and prevents the USB cable from being plugged in fully.

I had ten boards assembled with this connector and we have to shave some material off the curved outside of the headphones to get cables plugged in properly:

[image]

nmschulte, 2025-02-10:
> The USB-C connector added in [fa917dc](https://github.com/jamesturton/bose-qc35-usb-c/commit/fa917dc3a2d1e88145c8b8cd6d98b37e5f1b366f) (Molex 2171790001) connects too deep and prevents the USB cable from being plugged in fully.

Any more detail as to why fa917dc was performed to begin with?
> Use alternative USB C port which doesn't require modification

The new one seems to have issues with certain cables, but the old one required physical modification itself (vs scalping the headset case)?

What's a good part to use?

jamesturton, 2025-02-11:
> Any more detail as to why [fa917dc](https://github.com/jamesturton/bose-qc35-usb-c/commit/fa917dc3a2d1e88145c8b8cd6d98b37e5f1b366f) was performed to begin with?

The old connector has plastic mounting pins which need to be cut off manually as there is not enough space on the PCB to have these holes. This prevents the board from being assembled by pick-and-place. If you are hand assembling then it's not too big of an issue if you have a scalpel. It seems like most people I talk don't have the ambition to hand assemble the boards and would rather pay for a fab house to make the assembly for them so this is the biggest reason. The new part is also more widely available, rather than just being able to purchase from china.

> The new one seems to have issues with certain cables, but the old one required physical modification itself (vs scalping the headset case)?
> 
> What's a good part to use?

I've double checked the CAD models and the new connector should have the same insertion depth when accounting for the mounting position. @Kartoffel Have you tried multiple cables or just one? Where did you buy your cables from? It's possible cheap usb-c cables don't follow the usb spec on insertion depth and try to make them cheaper by using less material.

Kartoffel, 2025-02-11:
> The old connector has plastic mounting pins which need to be cut off manually as there is not enough space on the PCB to have these holes. This prevents the board from being assembled by [...]

Good to know the rationale, thanks.

> @Kartoffel Have you tried multiple cables or just one? Where did you buy your cables from? It's possible cheap usb-c cables don't follow the usb spec on insertion depth and try to make them cheaper by using less material.

We've tried a bunch of cables, I had 10 boards assembled for me and friends and we all had to sand away at the slope on the headphone shell.

I have been comparing the two designs side-by-side:

[image]

The old (Hroparts) connector is shown on the left, the new (Molex) connector is right.

The Molex plug keeps the cable molding about 2mm away from the board edge (indicated by the single line)
Do you know what this distance is for the old connector? I can't find it in the datasheet, but I suspect it's a bit further away as the connector itself hangs off the board by about 1mm.

jamesturton, 2025-02-11:
I have taken these dimensions from the CAD models supplied by the manufacturers

[image]

[image]

You can see that the inserting jack should mate 0.4mm further out from the reference plane on the back on the molex connector, or at least that's how I interpreted it.

The reality however matches what you have reported, I have take a photo of both connectors using identical cables and they do indeed seem to have the same sick-out, thus the molex should be moved 0.8mm closer to the edge of the board.

[image]

It seems there is only ~0.5mm between the pad to the edge of the board so I guess I'll have to find a new connector to use which has more overhang! If you have any suggestions then please let me know!

[image]

Kartoffel, 2025-02-14:
Thank you for investigating. I am not sure that 0.8mm is going to make a world of difference, we had to shave a fair bit more off the headphone housing to make plugs fully insert and latch.

One person even replaced the Molex connector by the original, and while it did make some difference still had to remove material around the jack. Did you not have to remove any? (Apart from enlarging the hole from micro-USB to USB-C)