jamesturton/bose-qc35-usb-c #20: KiCad v8 ?

nmschulte, 2025-02-11:
With my inexperience, attempting to update to KiCad v8 ... open the PCB and choose "Update PCB from Schematic" ...

> Warning: No net found for component J3 pad MP (no pin MP in symbol).
> Warning: No net found for component J3 pad MP (no pin MP in symbol).
> Error: J1 pad A not found in Connector_Molex:Molex_SlimStack_502426-2010_2x10_P0.40mm_Vertical.
> Error: J1 pad B not found in Connector_Molex:Molex_SlimStack_502426-2010_2x10_P0.40mm_Vertical.
> Total warnings: 2, errors: 2.

before:
[image]

after:
[image]

You can see that a lot of footprints change, and even some pins/pads.

----

Is a KiCad v8 upgraded project available?  I guess v6 is chosen for compatibility reasons, but unsure.

jamesturton, 2025-02-11:
I can update the project to using kicad v8, or even v9 which should be released in the coming weeks. I'll also update the USB connector used to one which doesn't require the case to be modified so violently!

nmschulte, 2025-02-12:
It seems most of my issue comes from the upgrade process: KiCad chose built-in libraries for the footprints that had matching names.  Switching everything to use `library.pretty/` (`library:` prefix) resolved this.

However, now there is a new contention: KiCad v8 DRC complains about SMD footprints with THT pins/pads, like the USB Type-C connectors shield.  This post explains it well, and the following tells about the "heatsink pad" "fabrication property", but it seems there's no "correct" solution yet -- just ignore this DRC specific.
- https://forum.kicad.info/t/footprint-component-type-doesnt-match-footprint-pads-expected-through-hole-actual-smd/51860
- https://gitlab.com/kicad/libraries/kicad-footprints/-/issues/2634